jlpteaching/dinocpu #35: Add ebreak and ecall instructions

powerjg, 2019-04-12:
These instructions raise a breakpoint exception for the debugger and make a request to the execution environment by raising an Environment Call exception.

We might be able to integrate this into the #34 CSR unit since it's the same opcode.

This would also make adding more workloads #29 easier, and potentially allow us to make a better host-guest interface.

Implementing `ecall` would also allow us to get printf working!

nganjehloo, 2019-04-19:
I will take care of this :)

nganjehloo, 2019-05-14:
Also finished this, and added mret to return from traps as well. only wait for interrupt is not implemented, but can be defined as a nop according to the risc-v standard

powerjg, 2019-05-14:
Awesome! Close these once you merge the changes :).

closed 2019-05-17 by nganjehloo