jlpteaching/dinocpu #40: Minor bug with included synchronous memory write test

deyahruhd, 2019-04-23:
While writing up the asynchronous memory tests, I noticed in the normal memory unit test the ["store and load words" test is using the MemoryUnitReadTester](https://github.com/jlpteaching/dinocpu/blob/17d7ca28141422a4e4a93ce92846e11263d0dcc9/src/test/scala/components/MemoryUnitTest.scala#L96) when it seems that it should be using MemoryUnitWriteTester. 

I corrected this in deyahruhd/dinocpu@55fbcc9e55d7b29dbaf8b33ac3b0584c5a268633 but the test failed when executed:

```
[info] [0.000] SEED 1555963536079
[info] [0.012] EXPECT AT 257   io_imem_instruction got 355 expected 100 FAIL
[info] [0.012] EXPECT AT 257   io_dmem_readdata got 355 expected 100 FAIL
[info] [0.013] EXPECT AT 258   io_imem_instruction got 355 expected 101 FAIL
[info] [0.013] EXPECT AT 258   io_dmem_readdata got 355 expected 101 FAIL
[info] [0.013] EXPECT AT 259   io_imem_instruction got 355 expected 102 FAIL
[info] [0.013] EXPECT AT 259   io_dmem_readdata got 355 expected 102 FAIL
...
[info] [0.060] EXPECT AT 766   io_imem_instruction got 355 expected 609 FAIL
[info] [0.060] EXPECT AT 766   io_dmem_readdata got 355 expected 609 FAIL
[info] [0.060] EXPECT AT 767   io_imem_instruction got 355 expected 610 FAIL
[info] [0.060] EXPECT AT 767   io_dmem_readdata got 355 expected 610 FAIL
[info] [0.060] EXPECT AT 768   io_imem_instruction got 355 expected 611 FAIL
[info] [0.060] EXPECT AT 768   io_dmem_readdata got 355 expected 611 FAIL
test DualPortedMemory Success: 2 tests passed in 773 cycles in 0.061594 seconds 12549.95 Hz
[info] [0.060] RAN 768 CYCLES FAILED FIRST AT CYCLE 257
[info] MemoryTester:
[info] DualPortedMemory
[info] - should have all zeros (with treadle)
[info] DualPortedMemory
[info] - should have increasing words (with treadle)
[info] DualPortedMemory
[info] - should store and load words (with treadle) *** FAILED ***
[info]   false was not true (MemoryUnitTest.scala:95)
```

It couldn't possibly be the memory module or else we would've had lots of problems involving writing to memory during WQ19, so it must be the unit test itself. Right now I'm working on fixing it but I didn't expect the test to just flat out fail, which is why I'm writing up this issue in advance.

deyahruhd, 2019-04-23:
Fixed it in deyahruhd/dinocpu@7ef8a17e326416cd5a3059fdc8dd5ce1740885b8. I found out that if a wire is poked with a certain value it retains that value for the rest of the test until it's updated with something new. 

After the write portion of the test memwrite was set to 1 and writedata to 355, meaning that on the read test the test was simultaneously reading and overwriting the values in memory to 355, which explains why the majority of the tests fail by encountering a value of 355. Poking memwrite to be false before the reads fixes the issue. 

I'll include the commits for this bug fix in the PR for the asynchronous memory module once I've verified that it works.

powerjg, 2019-04-23:
The fix looks good to me. You could remove some of the extraneous changes (e.g., changing `i` to `j`), but it's not a big deal either way.

BTW, try not to combine multiple "fixes" into a single PR. It'll be easier to track things and review things if each fix has its own PR. This is a very small fix, so I could see it going either way. However, I like to err on the side of more PRs.

PR #41 (2019-04-25): Fix memory write tests
As per what was advised in the issue for this pull request I reverted the change to the variable name. This PR closes #40
closed 2019-04-25 by powerjg