jlpteaching/dinocpu #54: Bug in reading bytes/halfwords

powerjg, 2019-05-13:
From @cjnitta:

Jason,
A student pointed out that there might be a bug in the memory.scala. 
Starting on line 73, it might need to be:
     when (maskmode =/= 2.U) { // When not loading a whole word
       val offset = io.dmem.address(1,0)

       when (maskmode === 0.U) { // Reading a byte
         readdata := (memory(io.dmem.address >> 2) >> (offset * 8.U)) & 
0xff.U
       } .otherwise {
         readdata := (memory(io.dmem.address >> 2) >> ((offset & 0x2.U) 
* 8.U)) & 0xffff.U
       }
     } .otherwise {
       readdata := memory(io.dmem.address >> 2)
     }
The difference was the shifting down of the byte or half word. It didn't 
seem to be a problem in the first 3 projects so far.

Chris

---

 I think both lines could be ` >> (offset * 8.U)` since the halfword must be 16-bit aligned.

powerjg, 2019-05-13:
@deyahruhd. This is a bug in your code, too :).

powerjg, 2019-09-03:
Does anyone remember if this has been fixed?

deyahruhd, 2019-09-06:
This should be fixed in the memory port modules.

closed 2019-09-06 by deyahruhd