jlpteaching/dinocpu #58: Travis CI fails builds due to java.lang.NoClassDefFoundError

deyahruhd, 2019-05-14:
Seems like every PR is going to fail the Travis CI check now due to this error occurring in every single unit test. [Here's an example with my asynchronous memory PR.](https://travis-ci.com/jlpteaching/dinocpu/builds/111676897) Some kind of misconfigured dependency, maybe?

It only started happening after #55 was merged into master, so @DanG100 could you look at this?

DanG100, 2019-05-14:
There is some incompatibility with the CI caches due to changing to a snapshot version, I manually cleared the cache and reran the tests. This should only happen once after merging #55, but I'll leave the issue open until that's confirmed.

powerjg, 2019-05-15:
Looks like this is resolved now?

DanG100, 2019-05-16:
Manual fix for this issue, go to https://travis-ci.com/ and select the dinocpu repo. Click the last menu option on the right (3 lines button). Open the caches menu and delete the cache for the failed build. Then rerun the build. This shouldn't happen now that all branches are ahead of 169ced5f2c1705cf2a0df24a6f4d1ce99c84d23e

powerjg, 2019-05-16:
Cool. I'll close this now, but we can reopen if it comes up again.

closed 2019-05-16 by powerjg