jlpteaching/dinocpu #84: Fix Treadle output directory bug

deyahruhd, 2019-08-08:
At the moment, setting the output directory of the simulators results in compilation errors due to a bug within Treadle. A temporary fix in #82 was to comment out the line which changed the output directory, but then this pollutes the current working directory

Unless this bug is ultimately intentional and we need to refactor the `Driver` code to instead use `FIRRTLMain` we will have to pay attention to when the bug is fixed by the Chisel guys.

seldridge, 2020-02-13:
This was mentioned on the FIRRTL gitter. I can see if I have some time to look at what's going on.

You're not supposed to have to change from Driver/OptionsManagers to Annotations (though you're encouraged). So, this is supposed to work as is. Sorry about breaking your stuff. :cry:

powerjg, 2020-02-13:
Hey Schuyler! Thanks for looking at this!

TBH, I'm happy to transition the code to the Annotations style. Could you point us to some documentation on that? I tried a few times to do everything via annotations, but I wasn't able to figure out enough context from looking at the FIRRTL/Chisel/Treadle code.

Also, don't judge my code style too much ;). I'm just a busy asst. prof. trying to throw something together for classes! (Plus, I don't know scala... or java... ;))

seldridge, 2020-02-13:
:wave:

This is the best documentation that I have right now: https://github.com/freechipsproject/www.chisel-lang.org/pull/39. Clearly this is incomplete, but the intent is to show a toy implementation of the new way to do things (independent of anything Chisel/FIRRTL related). I do have a draft (read: rejected) paper that gets into how this works abstractly which I can pass along to you directly if you want.

You may also be able to peruse the [`chisel3.stage` package](https://github.com/freechipsproject/chisel3/tree/master/src/main/scala/chisel3/stage) which is a rewrite of the Chisel3 `Driver` to use annotations.

And, *if there are any issues that you don't feel are getting traction, you can always reach out to me directly*.

Nonetheless, super excited that you're using DINO CPU in the UC Davis coursework. Would it be fine if I [add this project to the website](https://www.chisel-lang.org/community.html#projects-using-chiselfirrtl)?

powerjg, 2020-02-13:
Please do add it to that website. I had been meaning to reach out about that for a while, but kept getting distracted by other things. We also have a paper, if you want to link that (https://dl.acm.org/doi/10.1145/3338698.3338892)

I'll take a look at that documentation and the stage package. Hopefully, I'll get to this before the start of next quarter :).

Thanks for the info! And, of course, all of the hard work on Chisel/FIRRTL. It's a fantastic tool!

closed 2020-03-24 by powerjg