jlpteaching/dinocpu #91: Add a pipeline stage register interface and module

deyahruhd, 2019-10-16:
Right now, especially for the non-combinational pipeline, there are way too many notions of what "bubbling" does to a stage's register. With the combinational pipeline bubbling "pauses" or "freezes" IF/ID and flushes tbe control signals of ID/EX and EX/MEM. With the non-combinational pipeline, however, it is likely necessary to have to bubble, freeze, and flush every stage of the pipeline, so what bubbling actually does gets confusing.

I suggest to instead implement a new module that wraps these pipeline registers and provide a common interface for bubbling and flushing them. In particular, it should provide `valid`, `freeze`, and `flush` input signals, and a `good signal`, as well as `input` and `output` bundles for the actual contents of the registers.

The `valid` signal should tell the module to write the contents of the `input` bundle into the register. This is necessary for delayed memory, as a valid memory response is not immediately guaranteed and so we would have to watch for garbage input data.

The `freeze` signal should tell the module to ignore the `valid` signal. This has the effect of bubbling the register, similar to the IF/ID stage.

Lastly, the `flush` signal should have the module zero out the contents of the register on the next cycle. For compatibility this would zero out just the control signals of ID/EX and EX/MEM, but a simpler implementation would be to split these stage registers into two modules (one for control signals, the other for data signals). This approach also allows more granularity in the a bubbling operation: we could freeze the control signals, but zero out the data, or vice versa.

powerjg, 2019-10-17:
This mostly sounds great!

My only feedback is that we should try to have the simplest interface
possible. E.g., just a valid bit, bubble input, and freeze input. Or
whatever the *minimum* is.

Even if this means a little more complexity in the CPU, I think this
interface should be small.

deyahruhd, 2019-10-18:
One other comment I have, is it best for me to follow the factory interface patterns followed in Chisel? For example, if we consider the `Valid` interface, they provide a factory that generates a `Valid` from a supplied data bundle. That way, users can directly call `new Valid(mydata)` to add on a valid bit to their data.

Similarly, I think this new module should have a factory that automatically generates a `StageRegIO` from a data bundle. That way, we can simply replace the `RegInit(0.U.asTypeOf(new ...))` calls with `StageReg(new ....)`

powerjg, 2019-10-18:
Seems like a good idea to me!

deyahruhd, 2019-11-14:
Closed by #92.

closed 2019-11-14 by deyahruhd