jlpteaching/dinocpu #94: Move pipeline code to dinocpu.pipelined package

deyahruhd, 2019-11-13:
This should be simple to tackle - to make the dinocpu namespace cleaner, the pipelined CPU models and stage register should go into a `dinocpu.pipelined` package.

deyahruhd, 2019-11-15:
Closed by #95.

deyahruhd, 2019-11-16:
@powerjg, Looks like I need to do one more thing to close this issue. Since I moved the tester classes to `dinocpu.test`, all of the test commands in sbt must now utilize `dinocpu.test` instead of `dinocpu`: for example, `testOnly dinocpu.test.PipelinedCPUTester -- -z jal`. The documentation wasn't updated with this in mind since I forgot this would be an issue.

I'll create a new patch in a bit to update the docs.

powerjg, 2019-11-16:
Great, thanks!

deyahruhd, 2019-11-16:
Closed by #97. We could perhaps do some looking into a way to automatically alias the classes in `dinocpu.test` so that they can be run with `testOnly dinocpu.XXXXX`.

closed 2019-11-16 by deyahruhd