jlpteaching/dinocpu #110: build.sbt uses unresolved jelf dependency

deyahruhd, 2020-01-01:
I am not completely sure if this is an issue with my development setup. With IntelliJ IDEA Ultimate on Windows 10 and a freshly cloned copy of the repository I encounter errors about sbt not being able to resolve the jelf dependency:

```
[info] Updating ...
[warn]  module not found: net.fornwall.jelf#jelf;0.2
[warn] ==== local: tried
[warn]   C:\Users\jard\.ivy2\local\net.fornwall.jelf\jelf\0.2\ivys\ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/net/fornwall/jelf/jelf/0.2/jelf-0.2.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   C:\Users\jard\.sbt\preloaded\net.fornwall.jelf\jelf\0.2\ivys\ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:/C:/Users/jard/.sbt/preloaded/net/fornwall/jelf/jelf/0.2/jelf-0.2.pom
[warn] ==== sonatype-snapshots: tried
[warn]   https://oss.sonatype.org/content/repositories/snapshots/net/fornwall/jelf/jelf/0.2/jelf-0.2.pom
[warn] ==== sonatype-releases: tried
[warn]   https://oss.sonatype.org/content/repositories/releases/net/fornwall/jelf/jelf/0.2/jelf-0.2.pom
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  ::          UNRESOLVED DEPENDENCIES         ::
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  :: net.fornwall.jelf#jelf;0.2: not found
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]
[warn]  Note: Unresolved dependencies path:
[warn]          net.fornwall.jelf:jelf:0.2 (D:\dinocpu\build.sbt#L59-60)
[warn]            +- edu.ucdavis.cs:dinocpu_2.12:0.5
[error] sbt.librarymanagement.ResolveException: unresolved dependency: net.fornwall.jelf#jelf;0.2: not found
[error]         at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:332)
[error]         at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:208)
...
```

So it is impossible for me, using the current build configuration, to run any tests or code unless I manually change line 59 in build.sbt and bump jelf to version 0.4.1 instead of 0.2. With this change everything appears to run fine.

 It doesn't seem to be a problem with Jenkins though.

powerjg, 2020-01-02:
I don't see any reason not to bump the version, do you?

powerjg, 2020-01-02:
Closed by #111

closed 2020-01-02 by powerjg