jlpteaching/dinocpu #136: Update Chisel to 3.4

powerjg, 2020-12-10:
I tried updating to 3.3 (from 3.2) but received the following error.

```
 java.lang.NoClassDefFoundError: Could not initialize class treadle.stage.TreadleTesterPhase$
```

I think we will need to update how we are manually running Treadle to get this to work.

joisty, 2021-01-13:
I'm having the same error with Chisel 3.4, though I have the appropriate jar files.

[error] (run-main-0) java.lang.NoClassDefFoundError: treadle/stage/TreadleTesterPhase$
[error] java.lang.NoClassDefFoundError: treadle/stage/TreadleTesterPhase$

powerjg, 2021-01-13:
Not surprising. There are some updates that need to be made to support >3.2.

I don't know what needs to be updated right now, and I definitely don't know when I'll be able to get around to it. I'm happy to accept pull requests, though!

Unfortunately, some of the tests are failing right now... not sure when I'll get to that either :/.

itviewer, 2021-01-18:
@powerjg @joisty 
see https://github.com/jlpteaching/dinocpu/pull/139
for me, every things is ok.